nox771/i2c_t3 #5: 10-bit Slave Addressing?

brendanmatkin, 2016-09-06:
More of a feature request: 

Any plans to implement communication with devices with 10-bit slave addresses? i.e. Teensy as Master requesting data from 10-bit addressed slave devices. 

I took a look at the requestFrom/sendRequest methods to noodle with it myself but it's a little over my head..

Thanks!

nox771, 2016-09-07:
I don't currently have plans to support this as I have no 10-bit slaves that I can test with.  I'm not even sure where to find some.  

Actually SMBus, PMBus, proper ARB support, HS-mode, and 10-bit addressing are all features that are currently missing.  I did have plans to fix ARB, but ran out of time.

systronix, 2017-03-09:
What about ARB do you want to change/add?

brendanmatkin, 2017-03-09:
@nox771 sorry I didn't reply to this. 

I've been using attiny841-based sensors using [this](https://github.com/SpenceKonde/ATTinyCore) core and [this](https://github.com/orangkucing/WireS) library. That library more or less supports 10 bit addressing (the first part of the match is handled in the library/HW, then you manually check the rest via SW in the callback and set the N/ACK). If you are interested, I could mail you a sensor and some basic code to play with (if you don't mind waiting a month or two).

nox771, 2017-03-10:
@systronix - the ARBL resolution is basically broken.  At one point (just prior to T3.6 release) I had an idea of what needed to be done to fix it, but it would potentially require some significant changes to the ISR code.  It would mean a lot of regression testing.  I don't have time at the moment to work on it, but in the future I might try again.

@brendanmatkin - thanks for the offer, but I think if I added 10-bit I can use other T3's as 10-bit slaves (I would probably try to add it on both Master and Slave).  This is probably simpler than ARBL, but again I don't have time at the moment.  I'll keep this in mind for future updates.