nox771/i2c_t3 #6: Teensy Audio Board I2c_t3 issues

StevenJUlbrich, 2016-10-08:
Hello,

I am trying to use a Teensy 3.2 with Audio Board as a i2c slave
when adding the i2c_t3 library it throws errors - conflicting declaration 'i2c_t3 Wire'
 extern i2c_t3 Wire;

it is stated that replacing Wire.h with i2c_t3.h should be done in the Audio library.
./hardware/teensy/avr/libraries/Audio/control_sgtl5000.cpp
• ./hardware/teensy/avr/libraries/Audio/control_wm8731.cpp
• ./hardware/teensy/avr/libraries/Audio/control_cs4272.cpp
• ./hardware/teensy/avr/libraries/Audio/control_ak4558.cpp

What do you recommend?

nox771, 2016-10-11:
You cannot use the standard Wire library and i2c_t3 at the same time.  If you use i2c_t3 in your main program, then any library that uses the Wire library must also use it.  So the straightforward way is to change Wire.h to i2c_t3.h in the Audio library (or any other library using Wire.h).

If you are using Arduino (Teensyduino) to build your program, then there is a more tricky method that leverages the build system such that you don't need to modify the libraries.  It is described in these links:
https://forum.pjrc.com/threads/21680-New-I2C-library-for-Teensy3?p=79995&viewfull=1#post79995
https://forum.pjrc.com/threads/21680-New-I2C-library-for-Teensy3?p=80340&viewfull=1#post80340

nuket, 2017-04-25:
One easy way to do this when using PlatformIO is to ignore the Wire library by setting `lib_ignore` in `platformio.ini`, for example:

> [env:teensy35]
> platform = teensy
> board = teensy35
> framework = arduino
> **lib_ignore = Wire**

PlatformIO has some major advantages over the Arduino environment as it is, such as linting and autocomplete.