nox771/i2c_t3 #10: Arduino Uno:  i2c_t3.h missing

jonathanrjpereira, 2017-07-16:
```
#include "i2c_t3.h"  // I2C library

                                    ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.
```
Can you just include these files in your repository?

nox771, 2017-07-16:
??? i2c_t3.h is in the repo:
https://github.com/nox771/i2c_t3/blob/master/i2c_t3.h

"Error compiling for board Arduino/Genuino Uno."
This library doesn't work for AVR boards like the Uno.  It only works for Teensy 3.x boards (ARM).

tonton81, 2017-07-16:
lol... :)

jonathanrjpereira, 2017-07-16:
Damn. I have literally scrolled through the first 2 page results of Google for the MPU 9250. All of them have some sort of compilation error or are just incompatible with the Arduino UNO.

closed 2017-07-16 by jonathanrjpereira