nox771/i2c_t3 #11: Teensy interrupt "lag" on messages

Dogmans, 2017-09-13:
Using the basic_slave and basic_master examples the interrupts are only triggered for the previous message, not the current one.

I.e. 
master sends "#1"
master sends "#2" -> slave shows "#1" received
master sends "#3" -> slave shows "#2" received

Is this a known issue? I've been using Teensy LC and 3.5/3.6 and it seems to affect all of them.

Thanks

Will

Dogmans, 2017-09-14:
Debugging, seems to be that it triggers the handler on REPSTART, so looks like it is not seeing the stop condition at the end of the last message.

nox771, 2017-09-15:
Seems odd.  Something like that should have been obvious back when the examples were created.  But a lot of changes happened back then, so maybe something is off.  I'll look into it.

Dogmans, 2017-09-15:
Thanks, I went as far as looking at what was happening to the value of "flt" on the slave then thought I was probably out of my depth. My pullup resistors were 4K7 and I had a teensy LC connected to a 3.5, in case that makes a difference.

nox771, 2017-09-20:
I've been able to duplicate this.  It appears to be something wrong in the Slave ISR code.  I did change some stuff in there between when LC support got added and 3.5/3.6 support got added, so it might be related to that.  LC/3.5/3.6 added STOP bit detection (which was missing entirely from earlier devices), but they didn't really document it, so it's sort of a lot of trial-and-error.  It would probably not affect 3.0/3.1/3.2 devices.  I'll do some more trial-and-error and see if I can get it fixed.

nox771, 2017-09-20:
I pushed an update for this.  See if it works for you.  On my side I've tested it between a 3.6 and LC in both directions as Master/Slave and it seemed to work.

Dogmans, 2017-09-20:
Great, I'll give it a go in the next couple of days, thanks for picking this up.

Dogmans, 2017-09-25:
Just tried it out and it worked! Thanks for sorting.

closed 2017-09-25 by Dogmans