nox771/i2c_t3 #13: 9.2+ LAST BYTE TWICE

TechMasterJoe, 2017-09-21:
has something to do with the ftl change
9.2+ gets LAST BYTE TWICE

nox771, 2017-09-21:
Argh.  I'll take a look at this tonight.  So the bug is last byte is repeated?  I'm guessing this is on LC/3.5/3.6 slave?

TechMasterJoe, 2017-09-21:
3.6 tested
also on logic peek
I have a teensy 3.6 as a slave and master
it gets data from a pi as slave (wire1) writes to an array and the array is sent out as master (wire)
sending data seams to wait tell a 2nd command comes in
ex:
i get 4 byte packet 1 add:reg:reg:data on wire1 send data to wire0 and the data wont send tell packet 2 comes in

nox771, 2017-10-01:
For the record - this issue was resolved over email exchange.  There were two issues:
1. Slave ISR on LC/3.5/3.6 was not correctly detecting RepSTART
2. Nested Wire calls (eg. calling Wire inside Wire1 ISR) was not working correctly

Both of these issues are fixed in newest v9.4 release.

closed 2017-10-01 by nox771