nox771/i2c_t3 #15: slave_range configuration gives unexpected results

sidewaiise, 2017-12-05:
I've tried implementing your slave_range implementation, where I have 1 master device, and a slave with an address range of 0x67 to 0x99.

This seems good in principal, however when I try to run the code... it only seems to allow the address1 address to register. I've tried on 2 other adjacent addresses and there's no acknowledgement of those addresses (Error code 2).

Please advise how best to get this working. 

Thanks

sidewaiise, 2017-12-05:
I just figured out the solution.

The Address range was not correct!

The example gives a range from 0x08 to 0x77... however I thought the range extended beyond this.

So this may help anyone else trying to set up a slave-range application... use the range provided in the example (or a subset of).

Thanks

closed 2017-12-05 by sidewaiise

nox771, 2017-12-05:
It is correct that an address of 0x99 will not work as I2C address range is 7-bit, not 8-bit, so max is 0x7F.  Further per spec the addresses at the very top and bottom of the range are reserved and should not be used, which is why the example uses 0x08 to 0x77.  Any address within that range should be ok.

sidewaiise, 2017-12-05:
Thanks for Confirming Brian, all is working now. Great library! Cheers