nox771/i2c_t3 #16: TwoWire error when used with SparkfunHTU21D lib

jackassplus, 2018-02-17:
This is probably simple, but I'm attempting to use the HTU21D with 12c_t3.h
First error is:
```In file included from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/SparkFun_HTU21D/src/SparkFunHTU21D.cpp:28:0:
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/SparkFun_HTU21D/src/SparkFunHTU21D.h:54:14: error: 'TwoWire' has not been declared
   void begin(TwoWire &wirePort = Wire); //If user doesn't specificy then Wire will be used```

in SparkFunHTU21D.cpp
```
//#include <Wire.h>
#include <i2c_t3.h>
#include "SparkFunHTU21D.h"
```

in SparkFunHTU21D.h:
```
//#include <Wire.h>
#include <i2c_t3.h>
```

errors here:
```
//Start I2C communication
void HTU21D::begin(TwoWire &wirePort)
{
  _i2cPort = &wirePort; //Grab which port the user wants us to use
  
  _i2cPort->begin();
}
```

nox771, 2018-02-17:
i2c_t3 does not export a TwoWire class.  Try this - find/replace all "TwoWire" with "i2c_t3", so it will be like this:

SparkFunHTU21D.cpp
```
//#include <Wire.h>
#include <i2c_t3.h>
#include "SparkFunHTU21D.h"

...

//Start I2C communication
void HTU21D::begin(i2c_t3 &wirePort)
{
  _i2cPort = &wirePort; //Grab which port the user wants us to use
  _i2cPort->begin();
}
```

SparkFunHTU21D.h
```
//#include <Wire.h>
#include <i2c_t3.h>

...

class HTU21D {

public:
  HTU21D();

  //Public Functions
  void begin(i2c_t3 &wirePort = Wire); //If user doesn't specificy then Wire will be used
  float readHumidity(void);
  float readTemperature(void);
  void setResolution(byte resBits);

  byte readUserRegister(void);
  void writeUserRegister(byte val);

  //Public Variables

private:
  //Private Functions
  i2c_t3 *_i2cPort; //The generic connection to user's chosen I2C hardware
```

Main program (using **Wire**)
```
#include <i2c_t3.h>
#include "SparkFunHTU21D.h"

HTU21D sensor;

void setup() {
  sensor.begin();
}

void loop() {
  float humid = sensor.readHumidity();
  ...
```

Main program (using **Wire1**)
```
#include <i2c_t3.h>
#include "SparkFunHTU21D.h"

HTU21D sensor;

void setup() {
  sensor.begin(Wire1);
}

void loop() {
  float humid = sensor.readHumidity();
  ...
```

jackassplus, 2018-02-17:
Thanks! that did it.