nox771/i2c_t3 #20: I2C Master Issue

jrdlm, 2018-05-23:
After uploading the Basic_Master program to the Teensy 3.2 when I check it in the Serial Monitor it shows Send Failed. Is there any change required when the slave is also a Teensy 3.2?

nox771, 2018-05-24:
The example code should work on a Teensy 3.2.  You are using **basic_master** example on the Master device and **basic_slave** example on the Slave device, right?

If so, then check to make sure both devices are getting power, and that you have external pullups attached to both SDA and SCL (something like 2 to 5 kOhm resistor on each to 3.3V).  Also make sure the SDA/SCL are not crosswired.

jrdlm, 2018-05-24:
I tried what you suggested. I used 3.3 kOhm resistors and still found the error to persist. On further searching, I found an error in the basic_master code with the error being in the i2c_status I2C_NOT_ACQ. Is there any other way or any other method I should try using to get the i2c running between the 2 teensys

nox771, 2018-05-25:
I can try to setup a test later, but this is almost certainly a case of incorrect connectivity.  The library has been in use a long time and tested extensively.  I don't think I can attach an image here.  Send me an email (my address is given in the .cpp file), and I can email back an image of the connectivity to try.  

Or you can leave a comment on the PJRC forum thread below (login and do "Reply to Thread"), and I can post a reply with image there. 
https://forum.pjrc.com/threads/21680-New-I2C-library-for-Teensy3