nox771/i2c_t3 #23: Guidelines for using DMA vs interrupts

bmegli, 2018-08-19:
Are there some recommendations when to use DMA vs ISRs?

Precisely I mean the `setOpMode()` with:
- `I2C_OP_MODE_ISR` 
- `I2C_OP_MODE_DMA` 

Kind regards

nox771, 2018-08-20:
DMA mode can be useful to reduce interrupt calls, which could be useful if you have a lot of other interrupt activity already.  As an example see the logic analyzer images in this post (click to enlarge):
https://forum.pjrc.com/threads/21680-New-I2C-library-for-Teensy3?p=62411&viewfull=1#post62411
Notice the "I2C0 Interrupt" signal between the Interrupt Mode and DMA Mode plots.

That post is somewhat old, but at the bottom it does explain the workarounds involved in getting DMA mode to work.  The short version is DMA mode should only be used on a clean reliable connection, for instance a short wire connection to a I2C display.  In those cases the connection could be tuned to give a marginal performance benefit over the interrupt method.  

On long-wire, or error prone connections the DMA method could behave inconsistently.  In particular Slave NACKs and Timeouts won't behave optimally due to the way DMA hardware ignores bus events (again refer to above post).

Overall I would recommend using ISR unless you need absolute max performance (such as max refresh rate on I2C display), then use DMA.  Although it is only one line of code difference between them, so maybe try both out.

bmegli, 2018-08-20:
Thanks,

This answers even the questions I didn't yet know I wanted to ask!

Kind regards

closed 2018-08-20 by bmegli