nox771/i2c_t3 #24: setSCL()/setSDA()/pinConfigure() don't work properly

mbenkmann, 2018-11-01:
All 3 functions won't work if there are no pullups connected to the default pins. I guess that's because these functions think the bus is busy. If these functions are called before Wire.begin(), they shouldn't try to look at the bus. Note that https://www.pjrc.com/teensy/td_libs_Wire.html explicitly says that setSCL() and setSDA() can be used before Wire.begin(), so for people basing their code on that page that would be the natural way to write code.

nox771, 2018-11-08:
The hardware isn't going to function properly if the pullups aren't there regardless of the order you issue calls.  The pinConfigure() which is used by setSCL() and setSDA() will abort on bus busy (there is a low-level way to circumvent that, but it is a pointless exercise if pullups don't exist).

You can use getSCL() and getSDA() to verify pin changes have occurred.

Regarding setSCL() and setSDA() before begin(), the code is setup to utilize existing pin settings if they are not provided in the begin() call, so the code is already configured to operate in either order.

mbenkmann, 2018-11-08:
I can only tell you that using setSCL()/setSDA() did not work for me with my Teensy 3.2. Execution stopped at the setSCL()/setSDA() call and did not continue no matter how long I waited. The only way I could change the SCL/SDA from pins 19/18 to 16/17 was to use begin(I2C_MASTER, 0,I2C_PINS_16_17). It didn't matter which order I tried. I admit that I was just guessing that the missing pullups on 19/18 were the reason. Just to avoid confusion. Of course I have pullups on 16/17. But I shouldn't need pullups on 19/18 if I don't want to use these pins for i2c.

mbenkmann, 2018-11-09:
I'll try it but it might take a while till I get around to it. But just conceptually I would like it better if there was a special case for setSCL/setSDA being used before begin. Something like:
 
void setSCL(uint8_t pin) {
     if (begin not yet called) {
        i2c->currentSCL = pin;
     } else {
        pinConfigure_(i2c, bus, pin, i2c->currentSDA, i2c->currentPullup, 1); 
     }
}

This way, if any of the begin() functions is later called that does not specify a SCL pin, the one set before begin will be used. As far as I can tell there is no need to verify the validity of the pin if setSCL() is called before begin() because the check will be done when begin() is called.

nox771, 2018-12-02:
I uploaded a v11 version with a fix for this.  It seemed to work on my test case.  If you have further problems then let me know.

closed 2019-03-06 by nox771