nox771/i2c_t3 #25: Licensing question

ttlappalainen, 2018-11-12:
I have been using you library on my boat project. One of them may become commercial and then I will be in trouble with GPL. I changed my NMEA2000 library to MIT long time ago, when someone asked me to do that. How about your library, does it need to be GPL or could it be changed to MIT?

nox771, 2018-11-17:
Yes this is something I plan on doing.  The current license is not actually GPL, it is LGPL.  This was a carryover from when the v1 code was originally created (it borrowed from existing Teensyduino Wire source at the time).  However since then it has been heavily rewritten, so I'm not even sure anything original is left.  Others have asked this also.  I was planning on swapping the header for MIT or BSD or something similar to eliminate concerns.  I just haven't had time to look into this yet.  As soon as I can find some time I'll investigate.

FrankBoesing, 2018-12-01:
I've rarely read anything so bold.
What do you offer him for your demand?

mbenkmann, 2018-12-01:
> I've rarely read anything so bold.
> What do you offer him for your demand?

No one is demanding anything here. What are you talking about? I am OFFERING my help in performing a licensing change he says he has planned. What's your problem?

tonton81, 2018-12-01:
royalties :)

nox771, 2018-12-01:
Well I had this penciled in for next July, is that a problem?

Actually the simple truth is that I've been very busy elsewhere (day job, having food on table, roof overhead, etc), and this just had not made it to the top of the priority list yet.  But in the near term this situation is only going to get worse, so I'll push this fix within the next day or so, in order to close this.

mbenkmann, 2018-12-01:
As I've said, if you need any help with the boring mechanical parts of replacing the license, just tell me and I'll help. As for my project, I'll probably be finished planning end of December. If by that point i2c_t3's license has been changed, I'll use that. If not, I'll use a different i2c library (probably piconomix-fwlib).

mbenkmann, 2018-12-02:
You should add a LICENSE file so that github picks up on the license.