nox771/i2c_t3 #27: Wire.end() not implemented

utacc, 2019-02-26:
I'm trying to communicate to a hybrid I2C/SPI device, which occasionally needs to be powered down. I have it isolated on its very own I2C bus, with nothing else on it, specifically so I can drop the I2C lines (in tests, the I2C lines actually (partially) power the device if I2C isn't end()-ed, SDA and SCL then get turned into outputs, and then set LOW). 
In the Wire library, the interrupts get shut off, and the pins get unassigned. Should we do that manually with this library, or are there plans to implement end()?

closed 2019-02-27 by utacc

nox771, 2019-03-01:
Sorry, been a bit busy.  If you think it is generic enough and want to share, then you can upload the code or do a pull request and I can incorporate it into the main repo.

utacc, 2019-03-01:
Lol I don't remember how, unfortunately,  my git skills are very rusty. I
also think it should be tested. It works for me, in my application, but
that's a low bar.
I can send you a diff delta too.

utacc, 2019-03-01:
It should be generic enough, though. I wanted it to be a reasonable
facsimile of the same function from the original Wire library. It seems to
work as well as the original does, electrically.

nox771, 2019-03-06:
Sure whatever you have is ok.  You can email it direct to me if you want, my email is in the i2c_t3.h file.  It may take a while for me to add anything, but I'll try to get it done.