nox771/i2c_t3 #28: Teensy 4.0

cho934, 2019-09-04:
Hi,

I've just bought a Teensy 4.0. What I need to change to use the 3 i2c wires ? If I change only pinouts, is it enough ?
Could you please update the documentation ?

BR,

nox771, 2019-09-04:
i2c_t3 does not currently support Teensy 4.  At the moment you need to use Wire library for I2C on that device.

ttlappalainen, 2019-10-04:
Do you have plans and possible schedule for supporting Teensy 4?

nox771, 2019-10-05:
I have acquired some hardware.  When I get time I plan to look into it, but it is unlikely to be soon (probably a few months).

Richard-Gemmell, 2019-11-07:
Hi Everyone,

I've created an I2C library for the Teensy 4 that implements slave mode. See https://github.com/Richard-Gemmell/teensy4_i2c.

It doesn't support any other Teensies and I haven't attempted to make it backwards compatible with i2c_t3. It does contain a Wire wrapper though. 

I hope some of you will find it useful. Please feel free to use it or borrow the code if you prefer.

cheers,
Richard

markusrudolf, 2024-12-10:
> I have acquired some hardware. When I get time I plan to look into it, but it is unlikely to be soon (probably a few months).

Any updates on this?

nox771, 2024-12-12:
Sorry but an update on this library is not currently planned for Teensy4.  I did acquire hardware but never got the opportunity to focus on it.  Unfortunately in my brief look at it, it seemed to need a large rewrite.  However as someone linked above, others have created I2C libs for T4 already, so I would recommend investigating those.