nox771/i2c_t3 #30: I2C Slave Receive Buffer

ScottSWu, 2019-10-28:
Hi, I've been having trouble with Wire.available() and Wire.read() not reading from the beginning of the receive buffer.

My setup is a Teensy LC (master) and Teensy 3.2 (slave) connected via i2c pins 18/19. Both are grounded, and there are 3.3k ohm pullups to the 3.3V pin on the Teensy 3.2. Both are also connected over serial for uploading and debugging.

The master reads from serial and passes it along line by line over i2c.
```
// Buffer contains some text, ending with a newline
Wire.beginTransmission(SLAVE_ID);
Wire.write(buffer, buffer_len);
Wire.endTransmission(false);
```

The slave reads from i2c and, for debugging, just prints it back to serial.
```
if (Wire.available() > 0) {
  Serial.print(Wire.read());
}
```

Here's where the issue is: in order for Wire.available() to be greater than 0, each line sent must be longer than the previous, and the slave only receives the characters that come after. For example if the master sends
```
abcd
ab
abcdefg
```

The slave will print
```
abcd
fg
```

I'm guessing this is because when the slave receives a new transmission, it writes from the beginning of the rxBuffer and sets rxBufferLength, however the rxBufferIndex is not reset. When a new transmission is received and is shorter, it will return rxBufferLength - rxBufferIndex as some negative number.

There's not really an easy fix here:
* Setting rxBufferIndex to 0 would cause previous transmissions to be lost if not consumed.
* Turning rxBuffer into a ring buffer would cause issues for code using onReceive.
* rxBufferLength is used in many places for both master and slave.

Interestingly, this happens with the Arduino Wire library as well. I found a related issue but outdated workaround in https://forum.pjrc.com/threads/35500-Teensy-3-2-Master-and-Slave-I2C-communication-problem?p=113711&viewfull=1#post113711 .

nox771, 2019-10-28:
I don't have a setup to test this at the moment, but I suspect the problem is coming from this line:
`Wire.endTransmission(false);`

This doesn't issue a stop at the end of the transmission.  Repeated-start is generally used in the context of preparing to read from a slave.  From the slave perspective a series of transmits with no stop is interpreted as a single very long transmit.  Try changing to this:
`Wire.endTransmission();`

ScottSWu, 2019-10-29:
Hmm I tried both with no luck. Changing it did help while using onReceive.

Here's a more concrete repro: https://gist.github.com/ScottSWu/a14abe1a733af9d074c6fb1918c64c0d
[image]

nox771, 2019-10-29:
I'm not sure if this is the issue, but the way you have the slave code operating is not ideal.  Serial itself transfers and can be delayed by interrupts.  Printing character-by-character via Serial while looping on Wire could easily lose characters.  Here are a couple other things to try:

- In the example code there is a "basic_slave" example.  It shows how to use receiveEvent() callback to move the Wire data into a buffer.  Main loop then checks that and then sends a single Serial call.
- An alternate version of that would be to put the Serial.printf() command it uses directly into the Wire receiveEvent() callback.  As Serial command may be blocking that could give nested interrupts, but as I recall I tried that before and it did work.

Your code doesn't need the outgoing data side of the "basic_slave" example.  But maybe try a variant of the method in basic_slave and see if it gives a better result.

ScottSWu, 2019-11-11:
Hi, just wanted to follow up that onReceive() works great, especially with lots of data. Thanks for the advice on interrupts.