nox771/i2c_t3 #31: Is there a way to undo the effects of Begin?

hmaarrfk, 2019-10-30:
Is there a way to undo the effects of begin?

I would like to use the I2C pins as regular GPIO pins after using them as I2C pins.

Is this supported? If not, can you point me to what I would have to do to help out on that front?

Thanks!

nox771, 2019-10-31:
There is no "disconnect" routine built-in, but you can use one of the following to change the GPIO function back to normal: 
```
pinMode(pin#, INPUT);
pinMode(pin#, INPUT_PULLUP);
pinMode(pin#, OUTPUT);
```
After that it should be possiible to reconnect I2C using something like:
`Wire.pinConfigure(SCLpin#, SDApin#);`

hmaarrfk, 2019-10-31:
And I assume the driver works in polling mode? So I won't be triggering any interrupts after I use the the GPIO?

hmaarrfk, 2019-10-31:
Or I guess I would have to manually disconnect from all the interrupts

nox771, 2019-11-04:
I have not tested this scenario, but the I2C should not generate interrupts if the pins are disconnected from the GPIO.  If you need something more effective than you can disable I2C interrupts in I2C_C1_IICEN, and disable I2C clock in SIM_SCGCx registers (refer to begin() code).