nox771/i2c_t3 #33: Interrupts either not running and/or call wrong function.

WallyWalrick, 2020-09-10:
I'm using a Teensy 3.5 
`    Wire.begin();
    Wire.setClock(400000);
    Wire.onError(I2C_0_OnError);
    Wire.onTransmitDone(I2C_0_TransmitDone);
    Wire.onReqFromDone(I2C_0_ReceiveDone);`
The main loop writes to the slave, the I2C_0_TransmitDone reads from the slave and the I2C_0_ReceiveDone processes the data.

This works well when only using 1 I2C. When using Wire and Wire1 using I2C_1_***** functions, the transmit/receive work as intended but only I2C_1_OnError will get called for errors on both busses.
I got round that by using 1 onError function for both.
Then I tried 3 busses with Wire2 using I2C_2_****** for onTransmitDone/onReqFromDone as well. This time I2C_X_TransmitDone would be called correctly but only I2C_1_ReceiveDone would be called.

Is it some limitation with too many interrupts and timeouts?

nox771, 2020-09-15:
This is an interesting case.  I don't think I've tried the onError functions spanning multiple buses concurrently.  Does your application require backgrounding all the transfers?  If not, and you run multibus using foreground functions (endTransmission/requestFrom) do you still have problems?  Or perhaps try backgrounding only some buses?

I don't think I have a short solution for you.  It is possible there are some problems in the code regarding backgrounded multibus activity.  I'll have to study it.

WallyWalrick, 2020-09-15:
I worked out a solution based on the limitations.
3 transmit, 1 receive and 1 error interrupt functions.
I checked Wire.done() and a few sanity checks in the receive interrupt and it all works. It gives some delays occasionally (microseconds delay) on some bus calls but not enough to be an issue.
I need to use non-blocking routines on all 3 buses for this use case, if I have time I'll check to see if foreground functions give the same issues but likely won't be for a few weeks. I could change to purely polling but like not needing to worry about polling in this case.