nox771/i2c_t3 #35: Incorrect order of operations?

mbenkmann, 2021-06-02:
In `void i2c_t3::resetBus_(struct i2cStruct* i2c, uint8_t bus)` I see this:

    digitalWrite(scl,HIGH);
    pinMode(scl,OUTPUT);

Shouldn't these 2 lines be swapped?

tonton81, 2021-06-02:
on Teensy 3 at least, you can set it high before or after OUTPUT, because if they do it the other way around it will be LOW by default and affect something on the bus while switching back to HIGH from LOW :)

nox771, 2021-06-03:
Previous comment is correct.  The SCL/SDA buses operate with passive pullups.  So if they are not being driven low they will be at high voltage.  When using the I2C peripheral we cannot control the state of the lines directly (the peripheral controls it), but the assumption is that it is logic high (SCL/SDA are high between I2C sequences).  

So to transfer control to direct control (away from the peripheral), as used in resetBus(), the SCL is preset high before enabling as an output.  That is to prevent a driven-low glitch if it is done the other way around.  If this manifested as just another clock it would not be such a big deal (resetBus is outputting clocks anyway), but the timing of that glitch low period would be uncontrolled, which could cause a problem in itself.